PricelessToolkit/Droplet #2: What cable for Soil Sensors

lordratner, 2023-01-16:
Hi! I just got my droplet set up, and it is excellent! Very cool.

What cables do you use for extending the connection to the soil sensors? I don't know what the plug is called, so I don't know what to search for.

Thanks!

PricelessToolkit, 2023-01-16:
Hi, I'm very glad you liked it!! 
The droplet has ( JST 2.54mm pitch 3-pin ) connector, and for the sensor, it depends on your sensor :)  if you use a sensor like mine, it's JST 2.54mm pitch 3-pin too. You can buy a cable for (3d printers "End Stops" )  https://s.click.aliexpress.com/e/_DErXOBt  just look at the polarity and pin order!!!!!  "can be easily swapped if needed"  If it's not difficult, leave a review in the store :) Thanks!!

PricelessToolkit, 2023-01-16:
I forgot to say if the cable is long better to find a shielded cable.

closed 2023-01-18 by PricelessToolkit