PricelessToolkit/Droplet #3: Short vs Long button press

lordratner, 2023-01-19:
Current behavior: When you press and hold the top button, droplet_button_short_press comes on, then droplet_button_long_press comes on. 

This makes it difficult to have different automations for short and long presses, since the short press will trigger any time you long press. 

Instead, droplet_button_short_press should only come on when the button is released quickly. 

Thoughts?

lordratner, 2023-01-19:
From ESPHome:
on_click
This automation will be triggered when a button is pressed down for a time period of length min_length to max_length. Any click longer or shorter than this will not trigger the automation. The automation is therefore also triggered on the falling edge of the signal.

So I'd recommend something like this:

```
binary_sensor:
  - platform: gpio
    name: "${name} Button Short Press"
    pin:
      number: 36
    on_click:
    - min_length: 10ms
      max_length: 499ms
      then:
        - lambda: id(oled).turn_on();
        - delay: 20s
        - lambda: id(oled).turn_off();

  - platform: gpio
    name: "${name} Button Long Press"
    pin:
      number: 36
    on_click:
    - min_length: 500ms
      max_length: 1000ms
```

I think you could also combine it into one binary sensor per the documents, but then you wouldn't have two entities in HA.

PricelessToolkit, 2023-01-20:
Hi, yes, the idea is good, thanks, I will add it to the config, then people will choose what is convenient for them.
I made it that way because in my use case, I don't need 2 automation. it is more important for me to see the data on the OLED. and I didn't want to click to activate the screen and then click again to trigger the automation. May I know what kind of automation you use for this button?

lordratner, 2023-01-20:
For the long press I toggle my grow light. 

For the short press I want to toggle the fan.

After thinking about it, I think better timings would be 
    - min_length: 10ms
      max_length: 250ms

for the short press and 

    - min_length: 250ms
      max_length: 2000ms

for the long press.

closed 2023-04-20 by PricelessToolkit