PricelessToolkit/Droplet #4: A couple questions

lordratner, 2023-01-20:
I'll put these here since the discussion section is not activated for this repo. 

1. How do you keep a pump on in home assistant? On the droplet I can hold the pump button, but in HA the pump only stays on for a second or less. 

2. Would you consider a bulk watering routine? I have my droplet feeding a seedling trays with 50 plants each. I would like to water them by waiting for the moisture level to get down to a certain point, then trigger the pump for xx seconds to feed the entire tray. If I only do multiple short bursts of water, the plants closest to the droplet tube get all the water. 

Right now I plan to set the "desired moisture level" to 0% to disable the automatic pumping from droplet. Then in HA I will set an automation that will trigger when moisture gets below xx%, run the pump for xx seconds (probably 30 to start), wait xxx seconds, then check moisture levels again. 

It could be nice to have two watering modes, immediate and bulk:

Immediate mode: Current behavior, when moisture drops below set point, pump runs in short bursts until moisture is above set point. User inputs for each zone:

- desired moisture level

Bulk Mode: When moisture drops below set point, pump runs for xx seconds, then waits for a few minutes, then repeats until moisture is above a target level
User Inputs for each zone:

- Trigger moisture level
- pump duration
- Target Moisture level

Thoughts? I'm really enjoying this device, thank you for making it!

PricelessToolkit, 2023-01-21:
You can change pumps "ON Time" by changing "update_interval: 2s" in esphome config, or comment out  "Comparison Sens1,2,3,4,5 sections" and activate your pumps by Home Assistant automation.

PricelessToolkit, 2023-01-21:
I'll try to explain it differently

1. if you want to control the pumps from Home Assistant then  comment out "Comparison Sens1,2,3,4,5 sections"
but keep in mind if you restart the home assistant during the watering the pumps will not stop!

2. I think  is better to choose the "time" of saturation of the soil with water, then change update_interval: XXs
it's more reliable, it won't overflow

lordratner, 2023-01-21:
Ok I'll try that. What's the best way to add droplet to EspHome? I just added it as an integration in HA, but that doesn't let me update the configuration. Do I need to use a USB cable?

PricelessToolkit, 2023-01-21:
1. Create a new config in ESPhome
2. Compile it and Download the XXXX.bin file
3. Turn OFF your home WIFI (so that Droplet does not find where to connect)
4. Connect to "Droplet  Fallback Hotspot" via WIFI and upload the XXXX.bin file.
  https://github.com/PricelessToolkit/Droplet#first-time-setup

lordratner, 2023-01-21:
Ok great, got it connected. I'll start fiddling around see what I come up with. Thanks!

closed 2023-07-18 by PricelessToolkit