PricelessToolkit/Droplet #12: pillow not ofunt for fonts

stefancvetkovic, 2023-11-06:
Hello Im having first steps, and for some reason, I cant build esphome through CLI on my Windows Im getting an error:
Failed config

font: [source .\droplet.yml:56]

  Please update your pillow installation to at least 4.0.x. (pip install -U pillow).
  - file: arial.ttf
    id: font1
    size: 8
    
I have already installed pillow package with version
Name: Pillow
Version: 10.1.0

Do i need to downgrade to exact 4.0 version or? Do you know maybe?

Thanks

stefancvetkovic, 2023-11-06:
Also what is the default password for OTA update?

PricelessToolkit, 2023-11-06:
Hi,
https://github.com/PricelessToolkit/Droplet/blob/main/Config/ESPHome/droplet.yaml
```
ota:
  password: "bfec11234xsfea57e1b5d3b985cfe06c"
```
let me check it with the new ESPHome version.

PricelessToolkit, 2023-11-06:
With the latest ESPhome compiled without a problem, I can't say anything about version requirements.

```
INFO ESPHome 2023.10.6
INFO Reading configuration /config/esphome/droplet.yaml...
WARNING GPIO5 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing droplet (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.0.1
INFO Installing esphome/AsyncTCP-esphome @ 2.0.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.0.1 has been installed!
INFO AsyncTCP-esphome@2.0.1 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@3.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
Compiling .pioenvs/droplet/src/esphome/components/adc/adc_sensor.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/bmp280/bmp280.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/dallas/dallas_component.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/dallas/esp_one_wire.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/dht/dht.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/display/display.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/display/display_buffer.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/display/rect.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/font/font.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/homeassistant/sensor/homeassistant_sensor.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/i2c/i2c.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/i2c/i2c_bus_arduino.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/i2c/i2c_bus_esp_idf.cpp.o
Compiling .pioenvs/droplet/src/esphome/components/ledc/ledc_output.cpp.o
[... 145 lines cut ...]
Compiling .pioenvs/droplet/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pioenvs/droplet/FrameworkArduino/esp32-hal-time.c.o
Compiling .pioenvs/droplet/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pioenvs/droplet/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pioenvs/droplet/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pioenvs/droplet/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pioenvs/droplet/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pioenvs/droplet/FrameworkArduino/libb64/cdecode.c.o
Compiling .pioenvs/droplet/FrameworkArduino/libb64/cencode.c.o
Compiling .pioenvs/droplet/FrameworkArduino/main.cpp.o
Compiling .pioenvs/droplet/FrameworkArduino/stdlib_noniso.c.o
Compiling .pioenvs/droplet/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/droplet/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/droplet/libFrameworkArduino.a
Linking .pioenvs/droplet/firmware.elf
RAM:   [=         ]  12.7% (used 41504 bytes from 327680 bytes)
Flash: [=====     ]  51.5% (used 945377 bytes from 1835008 bytes)
Building .pioenvs/droplet/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/droplet/firmware.bin"], [".pioenvs/droplet/firmware.elf"])
Wrote 0xf8370 bytes to file /data/build/droplet/.pioenvs/droplet/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 60.78 seconds =========================
INFO Successfully compiled program.
```

closed 2024-04-05 by PricelessToolkit