PricelessToolkit/Droplet #13: Capacity Sensor that are verified to work

apolloFER, 2024-02-14:
I just setup my Droplet (second one is on the way). I bought the moisture sensors that were linked in the README. I've compared them to the ones that were available through pricelesstoolkit.com - HW-390 20210603. They're the same.

From what I've read, there's so much stuff around these moisture sensors - different chips, different components, missing timers and resistors. Mine should be "fine" but it's mostly not working.

Calibrated voltage range in the OS (0.99V to 2.5V) is never seen. I've overriden the sensor so it logs the raw voltage it sees and it's in the 0.7V-0.8V range. Fully submersed in water and in dry air - no change (stays at 0.75V +/- 0.03V). When in fully dry soil it increases to almost 0.9V. But the values jump around all the time (+/- 0.05V). The range is narrow. Tried attenuation set to auto as well.

Are there any sensors that are fully verified and that are known to work correctly with Droplet? Or is it a hit and miss until I found some that are working?

PricelessToolkit, 2024-02-14:
The primary issue with the sensors I encountered involves a 1MΩ resistor that is installed but lacks a proper connection, and reading values dropping too slowly, I installed 1MΩ resistors on the droplet board, even if you have no connection on the sensor it will work. The signal pin needs to be connected to ground through this resistor. If the sensor provides a range of "0.7-0.8V", it can be considered non-functional. Additionally, there are two more aspects to check: firstly, the diode installed may sometimes not be the original one but a random diode. Secondly, if you have multimeter compare the capacitor's value as it might be too small, or the 1MΩ resistor is too small, leading to the capacitor discharging too quickly. These sensors were produced by everyone who had a pnp machine :) so yeah its a hit and miss unfortunatly.

apolloFER, 2024-02-15:
Because of this, I would recommend you remove the Aliexpress link for the moisture sensors from the README.

These ones seemed promising to me: https://www.aliexpress.com/item/1005004895663124.html but after going through reviews I noticed some are OK and some were flipped. So not any better than the ones in Readme.

I've ordered these: https://www.aliexpress.com/item/32908693444.html and I'll report back if they're OK. They should have NE555 which is not a great option (TL555C/I is better).

PricelessToolkit, 2024-02-15:
OMG!!! thanks for letting me know.

PS: Should I make my own sample sensors?  but I can't produce with such low prices.

apolloFER, 2024-02-15:
I think the goal is to try to find sensors that are proved to be good. The PCB itself is mostly fine (unless you get the revision with missing 1MΩ GND connection). If you decide to make your own samples, I'd welcome it. But like you said, it's hard to produce them at such low prices. Even if you restock these in your shop (tested ones) it would help a lot.

ruimarinho, 2024-04-12:
@apolloFER any update on your quest to find a reliable moisture sensor :)?

apolloFER, 2024-04-13:
Yeah, I got [these](https://www.aliexpress.com/item/32908693444.html) ones and they have been working well. Those are the v1.2 but have the correct voltage regulator. A friend have me his v2.0 and that one was OK and working but don't know where he bought it.

Now I'm dealing with issues with pumps. I bought 15 of the [linked ones](https://www.aliexpress.com/item/1005005815962951.html) but they have very very poor reliability. 5 have already died (3 of them while I'm away for 3 weeks) so I'll need to find better ones (or a better seller).
Different ones I'm planning to try:

- https://www.aliexpress.com/item/1005005845034014.html
- https://www.aliexpress.com/item/1005005849610603.html
- https://www.aliexpress.com/item/1005005238191161.html
- https://www.aliexpress.com/item/1005006308119076.html
- https://www.aliexpress.com/item/1005005364015831.html
- https://www.aliexpress.com/item/1005005732870398.html

PricelessToolkit, 2024-04-13:
> Yeah, I got [these](https://www.aliexpress.com/item/32908693444.html) ones and they have been working well. Those are the v1.2 but have the correct voltage regulator. A friend gave me his v2.0 and that one was OK and working but don't know where he bought it.
> 
> Now I'm dealing with issues with pumps. I bought 15 of the [linked ones](https://www.aliexpress.com/item/1005005815962951.html) but they have very very poor reliability. 5 have already died (3 of them while I'm away for 3 weeks) so I'll need to find better ones (or a better seller). Different ones I'm planning to try:
> 
> * https://www.aliexpress.com/item/1005005845034014.html
> * https://www.aliexpress.com/item/1005005849610603.html
> * https://www.aliexpress.com/item/1005005238191161.html
> * https://www.aliexpress.com/item/1005006308119076.html
> * https://www.aliexpress.com/item/1005005364015831.html
> * https://www.aliexpress.com/item/1005005732870398.html

Thanks for the links,
are you sure your dead pump is for water? In the past, I bought 20 for water, but it turned out to be for air when I opened it. It didn't have a rubber seal, so water entered the motor and ruined it.

apolloFER, 2024-04-13:
> Thanks for the links, are you sure your dead pump is for water? In the past, I bought 20 for water, but it turned out to be for air when I opened it. It didn't have a rubber seal, so water entered the motor and ruined it.

To be honest, it's more than likely. The Aliexpress description said it's water. But at the same time I've noticed that JQB1523 pumps are usually quoted as air, while JSB1523 are quoted as water. And mine were JQB but quoted as water. I'll try the others that are not JQX models to see if they're better and I'll open up the ones that died when I get back home.
Did you try running pumps that are designed for 3V/3.7V at 5V? I found some that look like the pump you have in video (90 degrees between inlet/outlet ports) but are 3.7V.

PricelessToolkit, 2024-04-13:
> To be honest, it's more than likely. The Aliexpress description said it's water. But at the same time I've noticed that JQB1523 pumps are usually quoted as air, while JSB1523 are quoted as water. [...]

The label may indicate it's for water, but that's not necessarily the case :). I run them at 5 volts, and the pumps operate for a maximum of 6 seconds at a time. If you need to pump liters of water, they might burn out. Also, when inspecting pumps that have stopped working, please inform us if there is water damage or just burnout.

apolloFER, 2024-04-13:
All of them failed while they were off for a day or two (after they were working). I usually "flush" the system after 30min (to get air through the pumps). My guess is it's going to be water damage and not burnout but I'll report back.

I ordered a bunch of [CJWP12](https://www.aliexpress.com/item/1005005779638400.html) which seem to have backwater prevention feature (also available [here](https://www.aliexpress.com/item/1005005849610603.html) and [here](https://www.aliexpress.com/item/1005005858670203.html)). I'll use them for Droplets 3 and 4 and report back about them.

apolloFER, 2024-04-13:
I did some datasheet googling.

JQB1523 pumps (those linked in README) come in series 1 and series 2 varieties. [Series 1](https://en.symtek.com.tw/wp-content/uploads/sites/2/2019/10/TCS-JQB1523-series1-datasheet.pdf) is an air pump, Series 2 is a vacuum pump (medium not specified, not sure if it's water or not). 
[JSB1523](https://en.symtek.com.tw/wp-content/uploads/sites/2/2019/03/JSB1523008.pdf) pumps are water pumps. 

I also found the [datasheet](http://www.conjoin.com.cn/upfiles/products/202306/CJWP12-AA.pdf) for the CJWP12 pumps. Those are quieter, use less current, and have comparable flow (80mL/min) + also have backflow protection. 

Since JQB are non deterministic in which versions there are, it's better to recommend JSB1523 or CJWP12 pumps. I'll need to confirm if CJWP12 fit in the Droplet 3D case.

PricelessToolkit, 2024-04-13:
Thanks!! I updated the Pump link and uploaded the datasheet.

ruimarinho, 2024-05-13:
Any feedback so far @apolloFER ?

apolloFER, 2024-05-14:
Sort of. I wanted to wait until I can properly test the pumps but I can give an early update.

CJWP12 are OK in size and work as expected. Haven't tried them yet with actual watering. The reason for that is that they come with the wrong connector. They don't have JST XH 2.54 connector but something much smaller. I was able to resolder another wire with a XH 2.54 to test them out. I'm now waiting for a batch of XH 2.54 connectors in order to rewire all the pumps (22 of them) that I got.

Also, they have a smaller inlet/outlet diameter - 3.5mm. So 4mm hoses are out of question. Even 3mm hoses don't give me reassurance that they won't fall out when water pressure kicks in. I'll probably use some acrylic silicone to bond the hose.

apolloFER, 2024-06-25:
Update on the pumps.

I've replaced all the connectors with XH 2.54 so they're working now without problem. 

I'm using 3mm hoses but they're also not the tightest fit. So I added some aquarium sillicon to the hose/inlet connection. 

I've tested the pump to a height of around 1.5m and it was able to push the water without problem. 

Unlike some other pumps, the water doesn't "drain" away when the pumps stop working. Bust be that backwater protection in the pumps themselves. I'm using distilled water so shouldn't cause any issues.

So the pumps themselves are pretty good - probably a bit better than JSB1523 counterparts. But the connector replacement is probably too big of a hassle.

PricelessToolkit, 2024-06-25:
Thank you for the update! Yes, I also think that changing the connector is too big of a hassle. Unfortunately, I can't put $1000 into stocking good quality pumps. If I import them, the selling price will go up 2-3 times without considering my margin.

closed 2024-11-04 by PricelessToolkit