PricelessToolkit/Droplet #15: Can't flash via usb-uart

bernardovalente, 2024-04-06:
Hi,

I lost the encryption key for esphome when I lost the previous HA config. The droplet gets reconized on the new install but I can't give it access to integrate without this key so I'm trying to flash a new config but I'm having trouble with getting it recognized on my w11 desktop.

I'm using a usb to ttl adapter, tried moving the jumper on 3,3v and 5,v  and only connecting:

txd-rxd
rxd-txd
gnd-gnd
vdd-to nothing

and then I followed your guide to manual reflash, with and without external power to the board but it allways fails to open and connect

any tips? thank you

PricelessToolkit, 2024-04-06:
ESP32 has 3.3V Logic Level, so put your USB TTL adapter to 3.3V

Connect TTL adapter to droplet "Don't connect USB yet"
1 ##### USB TTL ######
txd-rxd
rxd-txd
gnd-gnd
vdd-to 3.3V "on pin header for OLED"

2 ############

Push and hold the "PROG button" then Connect the USB to the PC, when programming starts you can release the button.

bernardovalente, 2024-04-24:
I still can't connect to the board it says serial connection refused. 
I'm afraid perhaps I messed up the board when connecting both the ttl adapter and the external power or maybe the 5v from the adapter..

I'm going to try on a linux box maybe it's a w11 problem

PricelessToolkit, 2024-04-24:
maybe your TTL adapter just can't supply enough power in 3.3V mode.

bernardovalente, 2024-05-11:
So I changed the TTL adapter and managed to flash a modified droplet.yaml but it can't connect to wifi and only outputs this error continuosly:

`[19:23:04]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:23:04]invalid header: 0x00000000
[19:23:04]invalid header: 0x00000000
[19:23:04]invalid header: 0x00000000
[19:23:04]invalid header: 0x00000000
[19:23:04]invalid header: 0x00000000
[19:23:04]invalid header: 0x00000000
[19:23:04]invalid header: **0x00000000`**

any thoughts?

PricelessToolkit, 2024-05-11:
Hi, YAML is ok,  for programming use https://web.esphome.io/ and for testing use a power supply.

[image]

The problem could be due to several things
1. Wrong file format "legacy vs modern".
2. Programmed incorrectly.
3. Not enough power.
4.  Domaged with 5V but the probability is very very low.

bernardovalente, 2024-05-31:
I've tried a few configs but after flashing and reseting it, it outputs the same error via usb, also when powered via jack and with a couple of 5v usb chargers it never seems to connect to wifi.

I'm using web.esphome.io and choosing modern.

On your yaml you have:

```
esphome:
  name: ${name}
  name_add_mac_suffix: true
  project:
    name: pricelesstoolkit.droplet
    version: "1.0"
    
  platform: ESP32
  board: esp32dev
```

perhaps should be?:

```
esphome:
  name: droplet
  friendly_name: Droplet

esp32:
  board: esp32dev
  framework:
    type: xxxxxxxxxxx?
```

PricelessToolkit, 2024-05-31:
The default config working perfectly, it is not a config problem.

Can you try to flash the blink example into the board with Arduino IDE and select those configs for erasing flash completely?

[image]

And then upload this bin file with "web.esphome" it is my exact config.

[droplet-factory.zip](https://github.com/user-attachments/files/15518438/droplet-factory.zip)

The password is in secret.yaml
```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
```

if this does not work for you, [contact me](https://www.pricelesstoolkit.com/en/contact-us) don't forget to add the order number.

bernardovalente, 2024-06-29:
I'm trying the bare mininum example because it coudn't compile the blink one but I'm getting this error:

```
Sketch uses 266169 bytes (20%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20184 bytes (6%) of dynamic memory, leaving 307496 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.6
Serial port COM5
Connecting.......
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:c6:f7:22:87:bc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Compressed 19744 bytes to 13604...
Writing at 0x00001000... (100 %)
Wrote 19744 bytes (13604 compressed) at 0x00001000 in 0.4 seconds (effective 363.8 kbit/s)...
File  md5: 9fef1c1dc138cc0789ab3580d40b7b3e
Flash md5: ab5728dbeeb6fc465c6bbc16dad27590
MD5 of 0xFF is 5141cba2467248253063971ed8466710

A fatal error occurred: MD5 of file does not match data in flash!
Failed uploading: uploading error: exit status 2
```

[image]

PricelessToolkit, 2024-06-29:
Ok, let's fix that!

1. **Check Python Installation**: Open CMD and type `python`, then press Enter. If you see output like `Python 3.9....`, Python is installed.

2. **Install Python**: If not installed, download and install the latest Python version, and ensure to check "Add Python to PATH".

   [image]

3. **Install esptool**: Open CMD and type:

 ```sh
pip install esptool
```

Erase Flash: Run the following command, replacing COM4 with your port:
 ```sh
python -m esptool --chip esp32 --port COM4 --baud 115200 --after hard_reset erase_flash 
```
Write Flash Status: Run:

```sh
python -m esptool --chip esp32 --port COM4 write_flash_status --non-volatile 0
```
Disconnect and reconnect the power

This should fix the problem!

PricelessToolkit, 2024-06-29:
Then,  Skip the Arduino IDE part and try flashing ESPHome.

bernardovalente, 2024-06-30:
So I ran the commands and then flashed the droplet.bin, everything went fine but It still didn't connect when powered by the 5v barrel and it gives the same error in the logs when connected by usb. I also tried a basic esphome config and it still didn't work. 
Any thing else I can try?

> ```
> [19:23:04]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> [19:23:04]invalid header: 0x00000000
> [19:23:04]invalid header: 0x00000000
> [19:23:04]invalid header: 0x00000000 [...]

PricelessToolkit, 2024-06-30:
Unfortunately, the only solution in this case is to replace the ESP32 module. sorry!

PricelessToolkit, 2024-07-04:
"ESP32 ESP-WROOM-32" https://fr.aliexpress.com/item/32899182878.html

PricelessToolkit, 2024-07-04:
If you live in Europe, you can send it to me and I will repair it for free.

PricelessToolkit, 2024-07-04:
https://www.lcsc.com/product-detail/WiFi-Modules_Ai-Thinker-ESP32-S_C277944.html

bernardovalente, 2024-07-13:
I ordered an esp32 replacement that I'll try and replace it myself, it's arriving soon, but it's great to know that you might be able to do something to perhaps repair it, I might take you on that offer.

Awesome, thank you! I'll get back to you soon.

closed 2024-11-04 by PricelessToolkit