PricelessToolkit/Droplet #16: Water source

ruimarinho, 2024-04-13:
Hi,

I’m very interested in this project and I’m just waiting for restocking to help support it as well.

I’m wondering what’s the expected water source? A water bucket? I’d like to have this connected directly to a faucet but it would need some kind of pressure control. Any suggestions?

Thanks for the tips!

apolloFER, 2024-04-13:
I use it with a 7L water bottle. But I'm also planning to get one of those large [25L containers](https://cdn3.jysk.com/getimage/wd2.large/113092) (drill holes for hoses). No idea on connecting it directly to a faucet - you might be able to do it with the extension board and some relays (to prefill another large container when it empties).

PricelessToolkit, 2024-04-13:
Hi, you can use a bucket and valve like in toilets with a float to auto-refill.

ruimarinho, 2024-04-13:
Or what @PricelessToolkit said 🤣 away with precision level meters and just use a mechanical valve which is much safer!

closed 2024-04-16 by PricelessToolkit