PricelessToolkit/Droplet #17: Is it ok to use pump motors on 3v?

stefancvetkovic, 2024-05-12:
Hello, your aliexpress link points to water pumps on 3V, model A, but you said that motor pumps should be on 5v?

PricelessToolkit, 2024-05-12:
It is one link but in the shop, there are 2 versions, Model "B JSB1523" is 5V Model "A JQB1523" is 3V. Datasheet https://github.com/PricelessToolkit/Droplet/blob/main/Water_Pump_JSB1523008.pdf

stefancvetkovic, 2024-05-13:
Yes, I know that there are two models A and B, one operates on 5v and other on 3v.
Regarding the 3D case and your youtube demo video, I would say that I need model A, which operates on 3V?

But you said in readme.md file:
5x Micro Pump outputs "5v"

So as far I can see I have two options:
1. use model B (parallel water nozzles), but then I will need to edit just a little bit 3D model for the housing
**OR**
2. use model A and not to be worried because it operates on 3v?

Thanks,

PricelessToolkit, 2024-05-13:
you dont need to modify anything it is the same size. use 5v version.

stefancvetkovic, 2024-05-13:
so your recommendation is to use type B  water pump on 5v? 
Your design of enclosure is designs to support type A water pumps, nevertheless, I mount type B, and it works just fine, I will then just do another casing for this type of water pumps

PricelessToolkit, 2024-05-13:
[image]

There is no difference between Type A or B; when you screw the lid, it holds both Type A and B pumps.

stefancvetkovic, 2024-05-13:
Ok, I will let then leave it as is. Model B motors (parallel nozzles) on 5V. Thanks!

closed 2024-11-04 by PricelessToolkit