PricelessToolkit/Droplet #22: Pumps turning off immediately after turning on

avrabi19, 2024-12-31:
When I use the physical buttons they work fine but when I send a command through home assistant they immediately turn back off

avrabi19, 2024-12-31:
I am using the pumps used linked 5 volt

PricelessToolkit, 2025-01-01:
It is normal behavior with current yaml

```yaml

- platform: template
  name: Comparison Sens1
  internal: true
  update_interval: 2s   # Seconds to pump water then stops pumping and starts comparing sensors data, if needed it will pump again
  lambda: |-
    if (id(Soil1).state < id(desired_mois_value_1).state) {
      id(pump1).toggle();
    } else {
      id(pump1).turn_off();
    }
    return {};

```

avrabi19, 2025-01-01:
Oh ok thanks so I have to write my own yaml and set that in a trigger?

closed 2025-01-01 by avrabi19

PricelessToolkit, 2025-01-01:
If you want all the logic in HA, yes. In the current YAML, the on and off logic is defined in YAML. It takes the max level value from HA, compares it with the sensor value, and then decides if the pump needs to be on or off.

avrabi19, 2025-01-01:
Awesome! Thanks for the very fast help Loving the board. I will probably order a second one in a month or two