PricelessToolkit/Droplet #26: I cant connect it to HA reconnects every 4.30min

stefancvetkovic, 2025-05-17:
Hello,

I have strange behavior the board connects to my wifi, but recconects every 4min and 30 seconds.

DO you know why is that?

Also, I'm unable to connect it to my HA, saying Unable to connect to the ESPHome device. Make sure the device’s YAML configuration includes an `api` section.

[image]

here is my code:
https://pastebin.com/9MDnyXEV

PricelessToolkit, 2025-05-17:
Hi, 

1. You need to include the API https://esphome.io/components/api.html
2. When you connect it to HA, open the wireless log, and we will see what's going on.

stefancvetkovic, 2025-05-17:
1. I already have `api:`  part, but I added now 
```encryption:
  key: xxxx
```

2.But I can't connect it to the HA, that's the issue. Maybe firstly I can try to access it directly to captive portal?

PricelessToolkit, 2025-05-18:
If it has original firmware "not reflashed", then the current way is to configure it through the captive portal.

stefancvetkovic, 2025-05-18:
No, it doesn't have.
I flashed it a year ago, and it worked, and now it doesn't work (not being in production for several months)
Can I ship back to you this device and you can check it? Or something like that?
I presume that default firmware is the one that you posted in ESPHOME folder for configuration?
-

PricelessToolkit, 2025-05-21:
No need to resend, Just erase memory with Arduino IDE, check "Erase full memory before sketch upload," and upload a blank sketch. Then flash with ESPHome, and it will work.