PricelessToolkit/Droplet #27: Pump recommendation not matching screenshot & 3D model

nstrelow, 2025-06-23:
Some time ago I bought this very cool board and the pumps recommended in the description:

"Water Pump "Model B JSB1523" 5V "Connector pitch 2.54mm" - https://s.click.aliexpress.com/e/_ok8vbDj"

This pump has 2 tube connection that go out the bottom.

The screenshot show something looking like the other "Model A" that is 3.7V and has one tube coming out of the front of the pump.

[image]

I bought probably the wrong pump (did buy it probably 3 years back): "CM1135T-UL"

It's width is 21 mm, which seems to be the same as the pump linked on Aliexpress. With that pump the house does not fit perfectly, the gap seems too small.

I am wondering if there are any up to date pictures using the 2 tube connections at the bottom pump?

Are there any recommendations to hold the pump in place, if attached to a wall? Maybe hotglue them?

Thanks and have a great week 😀

PricelessToolkit, 2025-06-23:
Hi, I can't find the picture of the pump "CM1135T-UL", so I can't say anything about that. Normally, if you are using a 3D printed case and you tighten the bolts, it holds the pumps in place. If your pump is smaller, you can add some double-sided tape.

Have a nice weekend.

nstrelow, 2025-06-24:
The pump is slightly bigger, but should be the same height & width as the Model B.

I am curious, was there any reason for the shift from pump shown in the picture to the recommended one where the 2 pump outlets are on the same side?

Thanks in any case for this cool board and all the best with your other cool projects 😁